QuantumLeaps/qpc #1: portable examples, use a bsp for conio/curses

ensonic, 2016-09-16:
Sadly most examples are provided for dos and not fro posix. Couldn't most of them be made more portable by a simple conio/curses wrapper (just the features that are used in the example).

quantum-leaps, 2016-09-16:
Which examples do you mean? How exactly would you like them to be modified? Are you referring to the Win32 examples as "dos"?

closed 2016-09-16 by quantum-leaps

ensonic, 2016-09-16:
Yes, the win32 examples like calc. The book recommends to try them along with reading, but when using e.g. linux, this is not as straight forward.

quantum-leaps, 2016-09-17:
Unfortunately, preparing and maintaining a version of each and every example for all possible desktop platforms (currently Windows, Linux, and Mac OSX), is very time consuming. Especially that in order to test all these builds, you have to use different machines (or different virtual machines), which is difficult to automate. I hope you can understand that.

Now, the primary target audience for the PSiCC2 book and the QP frameworks is the embedded systems community. For better or worse, this community has chosen Windows as the primary platform. This is where all cross-development tools run. (Of course, there are some tools that run on Linux, some other run on OS-X, but these are exceptions rather than the norm.) Therefore, the most examples for QP are prepared for Windows. Again, hopefully this makes sense to you too.

--MMS

ensonic, 2016-09-19:
I understand for sure. Although when looking at maker communities (e.g. hackaday.com), I see no one using windows for embedded projects :)

quantum-leaps, 2016-09-19:
The maker community is significantly different than professional embedded systems developers. The QP framework tries to strike a balance in supporting both, but this very discussion is an example how difficult it is to make everybody happy. For makers, there is a specific adaptation of QP-nano for Arduino. The QP/C++ framework also comes with a specific adaptation to the Qt cross-platform framework. The QM modeling tool is released on Windows, Linux and OS-X. All of this costs tremendous amount of time and effort, yet it apparently is still not enough. But really, the examples like the Calculator (that started this whole discussion) are trivial to build on Linux. All you need is to adapt a Makefile....