QuantumLeaps/qpc #7: how to override an event and using guard to handle special cases while still use handle of parent in non handled cases

davidjonz2005, 2020-07-19:
Hi Miro,
 First of all I am sorry for that I have to send this to Sourceforge forum but I could not register in that website and it blocked my IP Address.

to demonstrate my issue I have to explain my situation in my project, I am programming a character LCD using QPC and AO with HSM modeling. I created a Main state and there are some state inside it. I have five keys UP,Down,Left,Right and OK key. I tried to handle all that keys in main state so not to make repetitive code and all events from these keys will be handled in Main state. However,In some cases when OK button pressed it must be handled by states inside Main state and used to transient to another state . For this purpose I have to check some conditions and if they are true I will transient and else I need it to be handled by Main state. I checked QM and uses guard for that purpose but I can not use handles in Main state and I have to repeat my code in Main state. Do you have any suggestion for that

Thank you for your great work

quantum-leaps, 2020-07-19:
I'm not sure that I understand your problem. Could you attach your QM model? The model could have repetitions, but it would be helpful just to see what's going on...
--MMS

davidjonz2005, 2020-07-21:
Thank you for Your quick response, This is my model
[image]

The main Idea is to handle key event in just one place inside "Top" state so to reduce repetition (that is HSM advantage over FSM).

my problem is in AlarmSetting State where I use a guard. In else part I intend to use default handle of key event in "Top" state so I place 
` status_ = Q_SUPER(&Feeder_FeedSetting);`
inside action code however it is not working :(. You may have a better suggestion or may be it is a good idea to add a feature for guard condition that if the guard is not fulfilled HSM use default handler.

quantum-leaps, 2020-07-21:
If you wish the OK event to "bubble up" from the AlarmSetting to superstates (FeedSetting / Top) simply **don't use the [else] guard**.
--MMS

davidjonz2005, 2020-07-21:
Thank you,that worked

closed 2020-07-21 by davidjonz2005