QuantumLeaps/qpc #11: What do you think about Boost.SML ?

icodk, 2021-04-13:
There are many state machine implementations and it is actually hard to find out which is the best.
I bought your book Practical UML statecharts In C/C++ a while ago and got a lot of insight into the subject.
I recently came a cross Boost.SML ( https://www.youtube.com/watch?v=Zb6xcd2as6o)
As far as I can see it is the most simple and performant state machine framework I know of.

https://boost-ext.github.io/sml/index.html

What do you think ?

lipi, 2021-07-10:
QPC has very low memory footprint, its main target is microcontrollers, which often only have a few KBs of RAM and code space is measured in tens of kilobytes.

quantum-leaps, 2021-07-10:
What do I think about Boost.SML or other similar approaches?

Frankly, I think that all such attempts are completely inferior to a good straightforward implementation like in QP/C or QP/C++ according to any criteria you might choose:

- code readability
- traceabilty and ease of debugging
- RAM footprint
- ROM footprint (code size + constants) 
- execution speed
- stack usage
- suitability for embedded systems
- MISRA compliance

I would really challenge anybody to demonstrate any criterion under which Boost.SML or any DSL for state machines would outperform QP.

Even the referenced video "C++Now 2019: Kris Jusiak “Rise of the State Machines” shows clearly that the "simple switch statement" outperforms all Boost libraries and other such complex solutions in speed and memory use. This is the basis of performance for QP state machine implementation. But QP improves on the "simple switch statement" in terms of the other criteria.

--MMS