QuantumLeaps/qpc #13: QF_PUBLISH not working from QHsm objects in a QSPY configuration

stefanschober, 2021-08-18:
I found sth, I'd like to call a slight inconvenience in the current implementation of QP/C.
I used the "examples/workstation/dpp-comp" implementation as a starting point to play with the orthogonal  components pattern. After enhancing the model and adding some functionalities, which also uses the QF_PUBLISH macro from within the components, I found that this only compiles without QSPY support.
As soon as QSPY is enabled, the compilation of the components (QHsm instances) fails with the error message, that the structure QHsm does not have a 'prio' element.
The implementation of QF_PUBLISH with QSPY enabled looks like this:
```
#define QF_PUBLISH(e_, sender_) \
        (QF_publish_((e_), (void const *)(sender_), (sender_)->prio))
```
The (sender_)->prio parameter is only available in classes derived from QActive, but not in those derived from QHsm.
I overcame this by adding a dummy 'prio' attribute to my component class and calling QF_PUBLISH with 'me' as the sender parameter instead of '&me->super'. Yet I feel that this is a bit cumbersome...

Regards
Stefan

quantum-leaps, 2021-08-18:
Hi Stefan,
The `prio` attribute is used for the [QS Local Filter](https://www.state-machine.com/qtools/qs.html#qs_local). This `prio` attribute is obviously missing in the "Orthogonal Components", which inherit only `QHsm` and not `QActive`. Your solution to add the "prio" attribute directly to the "Orthogonal Component" subclass sounds reasonable to me and is not such a big deal. If you wish, you can make the `prio` attribute conditional by surrounding it with #ifdef Q_SPY:

#ifdef Q_SPY
uint_fast8_t prio;
#endif

In the Spy configuration, the `prio` attribute gives you actually something: it allows you to selectively filter only the desired components, if you so desire. This is a very fine granularity control that the new QS Local Filter design affords, and the price for this (the `prio` attribute) seems quite reasonable to me. The previous design wouldn't allow you to achieve this.

--MMS

stefanschober, 2021-08-19:
Thanks Miro,

that leaves me with one last question.
I couldn't see an easy way to make a class attribute conditional, when creating the class definitions and state machines with QM. Is there a possibility I just didn't recognize?

Thanks
Stefan

quantum-leaps, 2021-08-19:
Hi Stefan,
Please see the ["Conditional Compilation" Section in the QM manual](https://www.state-machine.com/qm/ce_cond.html). As you will read there, conditional compilation is supported for the following model items: "attributes, operations, states, and transitions". Specifically, the declaration of the conditional `prio` attribute should look as follows (the name property):

`prio?def Q_SPY
`
In general, I highly recommend reading the provided documentation.

--MMS