QuantumLeaps/qpc #16: Your opinion on RTIC

mryndzionek, 2022-01-24:
I would like to ask about your opinion on [RTIC](https://rtic.rs/1.0/book/en/). To it seems that it's based on the same core ideas, but without explicit (hierarchical) state machines. Almost everything is done from [ISRs](https://rtic.rs/1.0/book/en/by-example/hardware_tasks.html). It also has [message passing](https://rtic.rs/1.0/book/en/by-example/message_passing.html) and a [timer queue](https://rtic.rs/1.0/book/en/by-example/monotonic.html).

The interesting part is [managing resources](https://rtic.rs/1.0/book/en/by-example/resources.html) and specifically [managing *shared* resources](https://rtic.rs/1.0/book/en/by-example/resources.html#shared-resources-and-lock). Instead of using mutexes/locks directly, the programmer just specifies which variables/resources are shared by which 'task' and RTIC computes optimal locking strategy and enforces the 'locking sections'.

Do you think adding such a mechanism to QK (QXK) would be a good idea?

quantum-leaps, 2022-05-04:
The links to RTIC don't work, so it is impossible to provide comments. This "issue" is closed.

closed 2022-05-04 by quantum-leaps

mryndzionek, 2022-05-04:
I've fixed the links.