QuantumLeaps/qpc #17: qpc start-stop_nucleo-h743zi missing

wangyanjunmsn, 2022-01-28:
there is no example start-stop_nucleo-h743zi as qpc documentation described.

quantum-leaps, 2022-01-28:
Please see:

qpcpp\examples\freertos\arm-cm\start-stop_nucleo-h743zi

But in general, starting and sopping AOs is **NOT** recommended. The start-stop example has been provided only because a commercial customer expressly demanded that, but it does not mean that they were right. It only means that as a paying customer they could make demands.
--MMS

wangyanjunmsn, 2022-01-29:
I don't think so. For example ,  in USB to LIN/K converter, the BUS can work in LIN mode or in K mode, but not at the same time. There are 2 AOs, one for LIN and another for K. User can start only one AO. When user want another operation mode, must stop the previous AO then start new AO.

quantum-leaps, 2022-01-29:
It seems that you just outlined your solution: Your USB AO should have two main "modes": the "LIN mode" and the "K mode".  In state machines, "modes" translate to "states". Of course, those main states can have substates, so there could be the whole hierarchical state machine nested in each. No need to stop and start AOs...
--MMS

closed 2022-01-30 by wangyanjunmsn