QuantumLeaps/qpc #21: can a state Q_TRAN to SPUER? like Blinky_on back to Blinky_idle in fllowing code

hehao9051, 2022-07-07:
/*..........................................................................*/
QState Blinky_idle(Blinky * const me, QEvt const * const e) {
    QState status;
    switch (e->sig) {
        case Q_ENTRY_SIG: {
            BSP_ledOff();
            status = Q_HANDLED();
            break;
        }
        case KEY_POWER_SIG: {
            status = Q_TRAN(&Blinky_off);
            break;
        }
        default: {
            status = Q_SUPER(&QHsm_top);
            break;
        }
    }
    return status;
}
/*..........................................................................*/
QState Blinky_off(Blinky * const me, QEvt const * const e) {
    QState status;
    switch (e->sig) {
        case Q_ENTRY_SIG: {
            BSP_ledOff();
            status = Q_HANDLED();
            break;
        }
     case KEY_POWER_SIG: {
            status = Q_TRAN(&Blinky_idle);
            break;
        }
        case TIMEOUT_SIG: {
            status = Q_TRAN(&Blinky_on);
            break;
        }
        default: {
            status = Q_SUPER(&Blinky_idle);
            break;
        }
    }
    return status;
}
/*..........................................................................*/
QState Blinky_on(Blinky * const me, QEvt const * const e) {
    QState status;
    switch (e->sig) {
        case Q_ENTRY_SIG: {
            BSP_ledOn();
            status = Q_HANDLED();
            break;
        }
     case KEY_POWER_SIG: {
            status = Q_TRAN(&Blinky_idle);
            break;
        }
        case TIMEOUT_SIG: {
            status = Q_TRAN(&Blinky_off);
            break;
        }
        default: {
            status = Q_SUPER(&QHsm_top);
            break;
        }
    }
    return status;
}

quantum-leaps, 2022-07-07:
Of course. A transition can target any state, including the superstate of the source state and the source state itself (transition to self). This question and any other question of this type is "answered" by the QHsmTst state machine (see Figure 2.11 in the [PSiCC2 book](https://www.state-machine.com/psicc2)).

[image]

I would also highly recommend that you watch the [video "State Machines Part-8: Semantics of Hierarchical State Machines"](https://youtu.be/NxV7JlU0-F4)

closed 2022-07-07 by quantum-leaps

hehao9051, 2022-07-08:
static QState Blinky_initial(Blinky * const me, void const * const par) {
    /*.${AOs::Blinky::SM::initial} */
    (void)par; /* unused parameter */
    QTimeEvt_armX(&me->timeEvt,
    BSP_TICKS_PER_SEC/2, BSP_TICKS_PER_SEC/2);

    QS_FUN_DICTIONARY(&Blinky_off);
    QS_FUN_DICTIONARY(&Blinky_on);

    return Q_TRAN(&Blinky_off);
}

/*.${AOs::Blinky::SM::off} .................................................*/
static QState Blinky_idle(Blinky * const me, QEvt const * const e) {
    QState status_;
    switch (e->sig) {
        /*.${AOs::Blinky::SM::off} */
        case Q_ENTRY_SIG: {
            PRINTF_S("%s\n", "Blinky_idle entry");
            status_ = Q_HANDLED();
            break;
        }
        case Q_INIT_SIG: {
            PRINTF_S("%s\n", "Blinky_idle init");
            status_ = Q_HANDLED();
            break;
        }
        /*.${AOs::Blinky::SM::off::TIMEOUT} */
        case TIMEOUT_SIG: {
            status_ = Q_TRAN(&Blinky_on);
            break;
        }
        default: {
            status_ = Q_SUPER(&QHsm_top);
            break;
        }
    }
    return status_;
}

/*.${AOs::Blinky::SM::on} ..................................................*/
static QState Blinky_on(Blinky * const me, QEvt const * const e) {
    QState status_;
    switch (e->sig) {
        /*.${AOs::Blinky::SM::on} */
        case Q_ENTRY_SIG: {
            BSP_ledOn();
            status_ = Q_HANDLED();
            break;
        }
        /*.${AOs::Blinky::SM::on::TIMEOUT} */
        case TIMEOUT_SIG: {
            status_ = Q_TRAN(&Blinky_off);
            break;
        }
        default: {
            status_ = Q_SUPER(&Blinky_idle);
            break;
        }
    }
    return status_;
}
/*.$enddef${AOs::Blinky} ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^*/

/*.${AOs::Blinky::SM::off} .................................................*/
static QState Blinky_off(Blinky * const me, QEvt const * const e) {
    QState status_;
    switch (e->sig) {
        /*.${AOs::Blinky::SM::off} */
        case Q_ENTRY_SIG: {
            BSP_ledOff();
            status_ = Q_HANDLED();
            break;
        }
        /*.${AOs::Blinky::SM::off::TIMEOUT} */
        case TIMEOUT_SIG: {
            status_ = Q_TRAN(&Blinky_idle);
            break;
        }
        default: {
            status_ = Q_SUPER(&Blinky_idle);
            break;
        }
    }
    return status_;
}

like this code, when Blinky_off Q_TRAN(&Blinky_idle)，in Blinky_idle state , there is no Q_ENTRY_SIG event ,only have Q_INIT_SIG, Is that right? @quantum-leaps

quantum-leaps, 2022-08-01:
It's OK not to provide Q_ENTRY_SIG, Q_EXIT_SIG, or Q_INIT_SIG in a state handler. If you don't have anything to do upon entry, exit, or nested initial transition you simply don't provide that. I hope this clarifies the situation.
--MME