QuantumLeaps/qpc #23: Why different v7.0.1 commits?

rexut, 2022-09-24:
.. and why is there a brand new main branch (master) with an complete different v7.0.1 commit?

[image]

`git log --oneline --graph 61084b9c ..5c010817`
```
* 5c010817 7.0.2
* 899b4ea9 7.0.2
* 283e1c3d Update README.md
* 6d0dc29a 7.0.1
* 61084b9c (tag: v7.0.1) 7.0.1
* e922fc13 7.0.1
* 862265e1 7.0.1
* 20e4b27d (tag: v7.0.1rc2) 7.0.1rc2
* 6ac2e923 (tag: v7.0.0) 7.0.0
```

`git diff --stat 61084b9c..6d0dc29a`
```
 .gitignore                                                                                      |    4 +-
 .gitmodules                                                                                     |    3 +
 3rd_party                                                                                       |    1 +
 LICENSES/LicenseRef-QL-dual.qlc                                                                 |   26 +
 LICENSES/QM-EVAL-QPC.qlc                                                                        |   25 +
 LICENSES/QP-Arduino_GPL_Exception.txt                                                           |   46 +
 LICENSES/QP-RasPi_GPL_Exception.txt                                                             |   48 +
 LICENSES/QP-mbed_GPL_Exception.txt                                                              |   42 +
 README.md                                                                                       |    7 +
 doxygen/images/github-star.jpg                                                                  |  Bin 0 -> 25587 bytes
 doxygen/make.bat                                                                                |  186 +-
 doxygen/rsm.bat                                                                                 |   88 +-
 doxygen/rsm_qpc.cfg                                                                             |   14 +-
 examples/arm-cm/blinky_efm32-slstk3401a/qk/gnu/flash.bat                                        |   96 +-
 examples/arm-cm/blinky_efm32-slstk3401a/qk/iar/blinky-qk.ewd                                    | 8426 ++++++++++++++++++++++++++++----------------------------
 examples/arm-cm/blinky_efm32-slstk3401a/qk/iar/blinky-qk.ewp                                    | 6368 +++++++++++++++++++++---------------------
 examples/arm-cm/blinky_efm32-slstk3401a/qk/iar/blinky-qk.eww                                    |   20 +-
 examples/arm-cm/blinky_efm32-slstk3401a/qk/iar/blinky-qk.icf                                    |   56 +-
 examples/arm-cm/blinky_efm32-slstk3401a/qv/gnu/flash.bat                                        |   96 +-
 examples/arm-cm/blinky_efm32-slstk3401a/qv/iar/blinky-qv.ewd                                    | 8918 +++++++++++++++++++++++++++++------------------------------
 examples/arm-cm/blinky_efm32-slstk3401a/qv/iar/blinky-qv.ewp                                    | 6644 ++++++++++++++++++++++----------------------
 examples/arm-cm/blinky_efm32-slstk3401a/qv/iar/blinky-qv.eww                                    |   20 +-
 examples/arm-cm/blinky_efm32-slstk3401a/qv/iar/blinky-qv.icf                                    |   56 +-
 examples/arm-cm/blinky_ek-tm4c123gxl/qk/gnu/flash.bat                                           |   64 +-
 examples/arm-cm/blinky_ek-tm4c123gxl/qk/iar/blinky-qk.ewd                                       | 8426 ++++++++++++++++++++++++++++----------------------------
 examples/arm-cm/blinky_ek-tm4c123gxl/qk/iar/blinky-qk.ewp                                       | 6346 +++++++++++++++++++++---------------------
 examples/arm-cm/blinky_ek-tm4c123gxl/qk/iar/blinky-qk.eww                                       |   20 +-
 examples/arm-cm/blinky_ek-tm4c123gxl/qk/iar/blinky-qk.icf                                       |   56 +-
 examples/arm-cm/blinky_ek-tm4c123gxl/qv/gnu/flash.bat                                           |   64 +-
 examples/arm-cm/blinky_ek-tm4c123gxl/qv/iar/blinky-qv.ewd                                       | 8924 +++++++++++++++++++++++++++++------------------------------
 examples/arm-cm/blinky_ek-tm4c123gxl/qv/iar/blinky-qv.ewp                                       | 6046 ++++++++++++++++++++--------------------
 examples/arm-cm/blinky_ek-tm4c123gxl/qv/iar/blinky-qv.eww                                       |   20 +-
 examples/arm-cm/blinky_ek-tm4c123gxl/qv/iar/blinky-qv.icf                                       |   56 +-
 examples/arm-cm/dpp_efm32-slstk3401a/qk/gnu/flash.bat                                           |   96 +-
 examples/arm-cm/dpp_efm32-slstk3401a/qk/iar/dpp-qk.ewd                                          | 8426 ++++++++++++++++++++++++++++----------------------------
 examples/arm-cm/dpp_efm32-slstk3401a/qk/iar/dpp-qk.ewp                                          | 7058 +++++++++++++++++++++++------------------------
 examples/arm-cm/dpp_efm32-slstk3401a/qk/iar/dpp-qk.eww                                          |   20 +-
 examples/arm-cm/dpp_efm32-slstk3401a/qk/iar/dpp-qk.icf                                          |   56 +-
 examples/arm-cm/dpp_efm32-slstk3401a/qv/gnu/flash.bat                                           |   96 +-
 examples/arm-cm/dpp_efm32-slstk3401a/qv/iar/dpp-qv.ewd                                          | 8426 ++++++++++++++++++++++++++++----------------------------
 examples/arm-cm/dpp_efm32-slstk3401a/qv/iar/dpp-qv.ewp                                          | 6374 +++++++++++++++++++++---------------------
 examples/arm-cm/dpp_efm32-slstk3401a/qv/iar/dpp-qv.eww                                          |   20 +-
 examples/arm-cm/dpp_efm32-slstk3401a/qv/iar/dpp-qv.icf                                          |   56 +-
 examples/arm-cm/dpp_efm32-slstk3401a/qview/qview-dpp.bat                                        |   16 +-
 examples/arm-cm/dpp_efm32-slstk3401a/qview/qview-dpp1.bat                                       |   16 +-
 examples/arm-cm/dpp_efm32-slstk3401a/qxk/gnu/flash.bat                                          |   96 +-
 examples/arm-cm/dpp_efm32-slstk3401a/qxk/iar/dpp-qxk.ewd                                        | 8426 ++++++++++++++++++++++++++++----------------------------
 examples/arm-cm/dpp_efm32-slstk3401a/qxk/iar/dpp-qxk.ewp                                        | 7088 +++++++++++++++++++++++------------------------
 examples/arm-cm/dpp_efm32-slstk3401a/qxk/iar/dpp-qxk.eww                                        |   20 +-
 examples/arm-cm/dpp_efm32-slstk3401a/qxk/iar/dpp-qxk.icf                                        |   56 +-
 examples/arm-cm/dpp_efm32-slstk3401a/win32-gui/dpp-gui.sln                                      |   46 +-
[... 264 lines cut ...]
 examples/zephyr/dpp/src/bsp.c                                                                   |   11 +-
 examples/zephyr/dpp/src/dpp.h                                                                   |    6 +-
 examples/zephyr/dpp/src/philo.c                                                                 |    6 +-
 examples/zephyr/dpp/src/table.c                                                                 |    6 +-
 ports/lint-plus/lin.bat                                                                         |  138 +-
 ports/win32-qutest/qp.sln                                                                       |   60 +-
 ports/win32-qutest/qp.vcxproj                                                                   |  416 +--
 ports/win32-qutest/qp.vcxproj.filters                                                           |  154 +-
 ports/win32-qv/qp.sln                                                                           |   84 +-
 ports/win32-qv/qp.vcxproj                                                                       |  806 +++---
 ports/win32-qv/qp.vcxproj.filters                                                               |  178 +-
 ports/win32/qp.sln                                                                              |   84 +-
 ports/win32/qp.vcxproj                                                                          |  804 +++---
 ports/win32/qp.vcxproj.filters                                                                  |  178 +-
 ports/zephyr/README.md                                                                          |    3 +
 zephyr/CMakeLists.txt                                                                           |   52 +
 zephyr/Kconfig                                                                                  |   21 +
[... cut ...]
 zephyr/module.yml                                                                               |    3 +
 {ports/zephyr => zephyr}/qep_port.h                                                             |    0
 {ports/zephyr => zephyr}/qf_port.c                                                              |    9 +-
 {ports/zephyr => zephyr}/qf_port.h                                                              |    0
 {ports/zephyr => zephyr}/qs_port.h                                                              |    0
 338 files changed, 328507 insertions(+), 328163 deletions(-)
```

quantum-leaps, 2022-09-25:
Thanks for your interest in this repo!

Several big changes happened at milestone 7.x. For example, the `3rd_party` directory has been deleted and replaced with an external submodule. Also, as most examples for git commands use the "master" branch, this was introduced as well. Probably not the best idea...

quantum-leaps, 2022-10-19:
The issue is closed with the comment above.

closed 2022-10-19 by quantum-leaps