QuantumLeaps/qpc #24: esp-idf port: QF_MAX_ACTIVE too large at 32

gh4emb, 2022-10-19:
Hello Miro,

Thanks again for adding the esp-idf port and examples to qpc. Currently, due to the esp-idf setting configMAX_PRIORITIES to 25, setting [QF_MAX_ACTIVE to 32 in qf_port.h](https://github.com/QuantumLeaps/qpc/blob/bd2f89a297e9945fc27ff82ffacb4a3191ebf496/ports/esp-idf/qf_port.h#L41) prevents the build from succeeding. 

I last tested QP/C using esp-idf v4.3.2 (tag v4.3.2, commit 8bf14a9).  In that version of the esp-idf port of FreeRTOS, [configMAX_PRIORITIES is set to 25](https://github.com/espressif/esp-idf/blob/8bf14a9238329954c7c5062eeeda569529aedf75/components/freertos/port/xtensa/include/freertos/FreeRTOSConfig.h#L182).

In the example-espidf-dpp branch of my fork of qpc, I had [decreased QF_MAX_ACTIVE to 16](https://github.com/gh4emb/qpc/blob/0941027a24cd4bd7aa77dd66f99ca0e0583d33da/ports/esp-idf/qf_port.h#L41), but I realize I never submitted those changes as a pull request.  My apologies.

[Side note: since esp-idf v4.3.2, espressif has changed the location of configMAX_PRIORITIES in newer versions of the esp-idf, but [it still appears set to 25](https://github.com/espressif/esp-idf/blob/8476ba1b7a4aeddf382fe9a80f3acc619b87942f/components/freertos/esp_additions/include/freertos/FreeRTOSConfig.h)]

quantum-leaps, 2022-10-19:
How about defining [QF_MAX_ACTIVE in qf_port.h](https://github.com/QuantumLeaps/qpc/blob/bd2f89a297e9945fc27ff82ffacb4a3191ebf496/ports/esp-idf/qf_port.h#L41) as configMAX_PRIORITIES. Would that work better? Could you perhaps check and post to this discussion thread?
Thanks a lot,
--MMS

gh4emb, 2022-10-19:
Yes, that is a better idea, and it does indeed build: 

`/* The maximum number of active objects in the application, see NOTE1 */
#define QF_MAX_ACTIVE         configMAX_PRIORITIES`

quantum-leaps, 2022-10-19:
OK, thanks. The modification is checked in. Please check [ports/esp-idf/qf_port.h](https://github.com/QuantumLeaps/qpc/blob/master/ports/esp-idf/qf_port.h).
--MMS

quantum-leaps, 2022-10-24:
Closed as fixed.

quantum-leaps, 2022-10-24:
Closed as fixed.

closed 2022-10-24 by quantum-leaps