QuantumLeaps/qpc #25: remove esp-idf port's use of QF_PTR_RANGE() in qf_port.c

gh4emb, 2022-12-20:
Hi Miro,

In version 7.1.3, to match the updates to the FreeRTOS port and enable successful esp-idf build, it seems:

https://github.com/QuantumLeaps/qpc/blob/b05a831d0ac96404496257a471f3e6b689f14e83/ports/esp-idf/qf_port.c#L671

should change to:

```
Q_ASSERT_ID(930, (me->start <= fb_next)
                             && (fb_next <= me->end));
```

and https://github.com/QuantumLeaps/qpc/blob/master/ports/esp-idf/qf_port.c#L618-L619

should change to:

`     Q_REQUIRE_ID(900, (me->nFree < me->nTot)
                       && (me->start <= b) && (b <= me->end));
`

quantum-leaps, 2022-12-20:
Thanks for pointing this out. The macro QF_PTR_RANGE_() has been removed in QP 7.1.3 (because the newer MISRA-C:2012 no longer frowns on pointer comparisons). But the esp-idf port has not been updated accordingly.
The problem has been fixed and pushed to the qpc repo. Please check.
--MMS

gh4emb, 2023-01-09:
Thanks, Miro--this works.  My apologies for the delayed response.

-Hal

quantum-leaps, 2023-01-09:
Good to hear. This issue is then closed as fixed.
--MMS

closed 2023-01-09 by quantum-leaps