QuantumLeaps/qpc #27: A simple task can use this framework?

MrFoxwho, 2023-04-10:
Hello Miro,

I'm a new embedded developer, I wanna ask whether QP could work in a schdduled task.

Like... the basic os will trigger my task each 1ms, and I want to port QP only for this task.

B&R
Fox

quantum-leaps, 2023-04-10:
Hi MrFoxwho,
Yes, of course, QP can run periodic tasks, and in fact, QP will do it more precisely than the standard RTOS task with a `delay()` function. You have two options to implement such periodic trigger:

1. Use a [time-event](https://www.state-machine.com/qpc/struct_q_time_evt.html), which is an event that your task (a.ka. "active object") can request to be delivered periodically.
2. Post an event (typically an immutable event) directly from the system clock tick ISR (running at 1ms).

The first option also requires a system tick ISR running at 1ms, which then calls [QTIMEEVT_TICK](https://www.state-machine.com/qpc/qf_8h.html#ad75daa93a2ba318ac1a5ed0aa431354b).

Almost every standard example (in the `qpc/examples` directory) uses time-events, so you can take a look at how to use one. For example, take a look at the "blinky" example, where a time-event is used to trigger the blinking of the LED.

--MMS

MrFoxwho, 2023-04-10:
您发给我的信件已经收到，我会尽快做出答复。Thanks for your letter. I will reponse to your mail as soon as I return. If you need immediate assistance, please call me on telephone numberBest wishes

MrFoxwho, 2023-04-10:
> Hi MrFoxwho, Yes, of course, QP can run periodic tasks, and in fact, QP will do it more precisely than the standard RTOS task with a `delay()` function. You have two options to implement such periodic trigger:
> 
> 1. Use a [time-event](https://www.state-machine.com/qpc/struct_q_time_evt.html), which is an event that your task (a.ka. "active object") can request to be delivered periodically.
> 2. Post an event (typically an immutable event) directly from the system clock tick ISR (running at 1ms).
> 
> The first option also requires a system tick ISR running at 1ms, which then calls [QTIMEEVT_TICK](https://www.state-machine.com/qpc/qf_8h.html#ad75daa93a2ba318ac1a5ed0aa431354b).
> 
> Almost every standard example (in the `qpc/examples` directory) uses time-events, so you can take a look at how to use one. For example, take a look at the ["blinky" example](https://github.com/QuantumLeaps/qpc/tree/master/examples/workstation/blinky), where a time-event is used to trigger the blinking of the LED. Please note that there are many versions of the "Blinky" example for various embedded boards.
> 
> --MMS

Hi Miro,

Sorry, I didn’t describe the problem correctly.
Actually, I want to “ port QP only into this task”. That means I  already have a periodic running task, and hope to port this framework into this task, and only into this task. I want make this periodic task works like a small os.

quantum-leaps, 2023-04-10:
Hi MrFoxwho,
While it might be possible to fit somehow the cooperative QV kernel somehow inside a traditional RTOS task (periodic or not), I would generally advise against such ugly hybrids.
Most likely, what you really want is to use a hierarchical state machine inside your already existing periodic task. The layered architecture of QP has been actually designed for this. Specifically, you can subclass just the [QHsm base class](https://www.state-machine.com/qpc/struct_q_hsm.html), without taking the whole QActive. Then you just dispatch events to the QHsm subclass in your periodic task. You can code your HSM either manually, or you can use the [QM modeling tool](https://www.state-machine.com/qm) and generate code automatically. All of this would work. As an example application that uses a passive QHsm subclass, please see [qpc\examples\workstation\qhsmtst](https://github.com/QuantumLeaps/qpc/tree/master/examples/workstation/qhsmtst). There is also a [video that explains the semantics of HSMs](https://youtu.be/NxV7JlU0-F4) with the QHsmTst example.
--MMS

MrFoxwho, 2023-04-10:
> Hi MrFoxwho,
> 
> While it might be possible to fit the cooperative QV kernel somehow inside a traditional RTOS task (periodic or not), I would generally advise against such ugly hybrids.
> 
> Most likely, what you really want is to use a hierarchical state machine inside your already existing periodic task. The layered architecture of QP has been actually designed for this. Specifically, you can subclass just the [QHsm base class](https://www.state-machine.com/qpc/struct_q_hsm.html), without taking the whole QActive. Then you just dispatch events to the QHsm subclass in your periodic task. You can code your HSM either manually, or you can use the [QM modeling tool](https://www.state-machine.com/qm) and generate code automatically. All of this would work. As an example application that uses a passive QHsm subclass, please see [qpc\examples\workstation\qhsmtst](https://github.com/QuantumLeaps/qpc/tree/master/examples/workstation/qhsmtst). There is also a [video that explains the semantics of HSMs](https://youtu.be/NxV7JlU0-F4) with the QHsmTst example.
> 
> --MMS     

Thank you, Miro!
Your answer really helps.

BRs
Fox