QuantumLeaps/qpc #29: Possible typo error in freertos port in v7.3.0

sicrisembay, 2023-08-10:
Hi,

I'd like to report a possible typo error bug in freeRTOS port.  There are two instances of these error.

Instance#1 (qf_port.c)
`bool QActive_post_(QActive * const me, QEvt const * const e,
                   uint_fast16_t const margin, void const * const sender)`

https://github.com/QuantumLeaps/qpc/blob/08b4a782c74df7e68fe2dfd59bb31ce54f5df11d/ports/freertos/qf_port.c#L230-L231

It should be a call to xQueueSendToBack.

Instance#2 (qf_port.c)
`void QActive_postLIFO_(QActive * const me, QEvt const * const e)`

https://github.com/QuantumLeaps/qpc/blob/08b4a782c74df7e68fe2dfd59bb31ce54f5df11d/ports/freertos/qf_port.c#L271-L272

It should be a call to xQueueSendToFront.

best regards,
Sicris

quantum-leaps, 2023-08-10:
Hi Sicris,
Good catch! The names of the FreeRTOS queue send operations appear to be flipped in the QP-FreeRTOS port. The problem has been corrected and has been already pushed to GitHub. Please check:
- https://github.com/QuantumLeaps/qpc/blob/master/ports/freertos/qf_port.c
- https://github.com/QuantumLeaps/qpc/blob/master/ports/esp-idf/qf_port.c

Thanks a lot for reporting!
--MMS

sicrisembay, 2023-08-10:
That's quick!  :)
Thank you.

best regards,
Sicris

closed 2023-08-10 by sicrisembay