QuantumLeaps/qpc #33: About Freertos #if ( configMAX_PRIORITIES < QF_MAX_ACTIVE ) Condition

su782710477, 2023-11-15:
configUSE_PORT_OPTIMISED_TASK_SELECTION can only be set to 1 when configMAX_PRIORITIES is less than or equal to 32.  It is very rare that a system requires more than 10 to 15 difference priorities as tasks that share a priority will time slice.
But this seems very common that Qf active more than 10-15, even more 32. I think the condition No very good.
At the same time，one rtos priorities to one qf active it is unreasonable in freertos

quantum-leaps, 2023-11-15:
The configuration constant configUSE_PORT_OPTIMISED_TASK_SELECTION is only defined inside the application's FreeRTOSConfig.h header file. This file is intended to be adjusted for every application at hand. So, if your particular application uses more than 32 AOs, then by all means, set configUSE_PORT_OPTIMISED_TASK_SELECTION to 0 in your FreeRTOSConfig.h.

In the next QP release 7.3.1, the QP ports to 3rd-party RTOSes will be augmented such that you will be able to explicitly set the thread priority for the underlying RTOS thread by using the Q_PRIO() macro. For example, you would be able to set the same RTOS thread priorities for multiple AOs.

--MMS

su782710477, 2023-11-16:
thank you. It will resolve my doubt,

quantum-leaps, 2024-02-02:
In QP release 7.3.1, the QP ports to 3rd-party RTOSes are augmented such that you will be able to explicitly set the thread priority for the underlying RTOS thread by using the Q_PRIO() macro. For example, you would be able to set the same RTOS thread priorities for multiple AOs.
--MMS

closed 2024-02-02 by quantum-leaps