QuantumLeaps/qpc #36: About "QHsm_isIn" function

su782710477, 2023-12-11:
In my understand,  "xx.temp.uint = ~xx.state.uint;" is inoder to checkt the safe function. 
but QHsm_isIn is using xx.temp.fun directly, I can't understrand. of course the it occur an exception in my program. (the program lose control, and reset). i think it maybe "xx.temp.uint = xx.state.uint" and then do ohters, rather than use "xx.temp.uint" directly

quantum-leaps, 2023-12-12:
Yes, the use of the QHsm::temp attribute for the Duplicate Inverse Storage has broken the implementation of QHsm_isIn(). Thank you for reporting this.

The QHsm_isIn() implementation has been corrected and [is already committed to the repo](https://github.com/QuantumLeaps/qpc/blob/master/src/qf/qep_hsm.c#L453).

--MMS

su782710477, 2023-12-12:
Aha I like Qp, but the qm.exe edit the code is Not very good. 
I often use Qm.exe generate Frame, and use Vs code Edit code and then copy the code to qm.exe, finally generate the code again.
it is very Cumbersome.

quantum-leaps, 2023-12-12:
Hi there,
First, I think that the internal text editors in QM aren't that bad. You just need to give it a try and edit the information inside QM. This is the simplest solution and will avoid all your problems with constantly updating files between QM and another text editor. VS Code is very slow compared to QM.

Only if you utterly despise QM, and absolutely cannot live with its editor (which sounds like a religious war to me), you should consider using a 3rd-party editor. But then, you should properly use the QM feature called **external files**, which is described here:

https://www.state-machine.com/qm/ce_file.html#ce_file_extern

--MMS

closed 2024-02-02 by quantum-leaps