QuantumLeaps/qpc #40: /ports/win32-qutest/qutest_port.c uses non-standard include 'conio.h'

sulemankm, 2024-02-15:
Trying to compile qutest LedBar example on MSys and there is no such file 'conio.h' in gcc.

stefanschober, 2024-02-17:
The MSys environment doesn't provide conio.h. You should use the UCRT64 or MINGW64 environment to try this.

quantum-leaps, 2024-02-17:
The support for console input (and thus the obsolete header file <contio.h>) is not necessary for the QUTest host executables and will be removed in the next upcoming QP release 7.3.3. Most of the time, the executables are launched in the background, anyway (so there is no keyboard input). This means that QUTest test fixtures should compile with MSys, or Cygwin. (They compile with the standard MinGW for Windows, and that's why there was no problem).

This issue with the obsolete <conio.h> header file affects the standard QP ports to Win32, which also use it. The real problem is that there is no real good replacement for this functionality (short of using the "ncurses" library.) For that reason, the console-related functionality in the QP win32 and win32-qv ports will be compiled conditionally, only when the macros QF_CONSOLE is defined. This would eliminate the <conio.h> problem for QP applications on Windows that don't need console (e.g., GUI applications with QWin or "headless" applications.) For console applications, users will have an option either to define QF_CONSOLE and use <conio.h>, or not define it and provide the console services some other way.

--MMS

closed 2024-03-01 by quantum-leaps