QuantumLeaps/qpc #42: qpc 8.0 - qf_actq.c::QTicker_trig_ does not respect Q_UNSAFE setttings

stefanschober, 2024-10-28:
`QTicker_trig_()`:
...
`    if (me->eQueue.frontEvt == (QEvt *)0) {

        static QEvt const tickEvt = QEVT_INITIALIZER(0);
        me->eQueue.frontEvt = &tickEvt; // deliver event directly
        --me->eQueue.nFree; // one less free event

        QACTIVE_EQUEUE_SIGNAL_(me); // signal the event queue
    }`

in `QActive_get_()`:
the assertion `Q_INVARIANT_INCRIT(411, Q_PTR2UINT_CAST_(e)
                            == (uintptr_t)~me->eQueue.frontEvt_dis)`
fails because `me->eQueue.frontEvt_dis` was not updated during `QTicker_trig_()`

Best regards
Stefan

stefanschober, 2024-10-28:
patch proposal
[qf_actq.diff.zip](https://github.com/user-attachments/files/17543722/qf_actq.diff.zip)

quantum-leaps, 2024-10-28:
Hi Stefan,
Thank you for the report. The issue with QTicker has been reproduced. The fix and specific tests will be added soon.
--MMS

quantum-leaps, 2024-10-30:
Fixed and pushed into the qpc and qpcpp repos.
--MMS

closed 2024-10-30 by quantum-leaps