QuantumLeaps/qpc #45: qpc 8.0.0 - assertion 801 in qf_actq.c triggered, when message queue is assigned to ticker object

stefanschober, 2024-11-10:
I have a setup, which provides a 5 entry wide message queue to the ticker object in `QACTIVE_START()`. With this setting the assertion 801 'qfactq.c - line 572' will trigger and kill the application.
The assertion reads `Q_REQUIRE_INCRIT(801, me->eQueue.nFree == 1U);`

Is it intentional, that this assertion only accepts exactly 1 free entry in the ticker's message queue?
In this case, I would be interested in learning the rationale behind that decision. My understanding now is, that the ticker object must not be provided with a message queue. Would that be correct?

Thanks
Stefan

quantum-leaps, 2024-11-10:
Yes, the assertion `Q_REQUIRE_INCRIT(801, me->eQueue.nFree == 1U);` is correct. This is because the QTicker active object does NOT use the event queue buffer at all, and it is a mistake to provide any queue buffer. I understand that the best way would be to enforce it by some assertion, but for practicality, QTicker reuses the QActive_start() operation (which is no longer "virtual"), so it cannot contain such an assertion. The second best option is to be very clear about it in the documentation, which will be updated.

Now, the lack of an event queue buffer does NOT mean that QTicker can be triggered only once per tick. It can be triggered up to 0xFF times. It just stores this information inside a counter (which resides in the repurposed eQueue.tail attribute). The queue buffer is not needed because QTicker does not really have a state machine (please see the "virtual" `QTicker_dispatch_()` operation).
 --MMS

quantum-leaps, 2024-11-22:
Closed as a documentation issue, which has been fixed.
--MMS

closed 2024-11-22 by quantum-leaps