QuantumLeaps/qpc #49: question about passing driver data with Q_NEW api

andy27182, 2025-03-05:
When my software needs to access the data of the same peripheral in multiple QActives, is it the best practice in the qp framework to use Q_NEW to pass these data? 
The data of a certain peripheral may be dozens of bytes in size. When there are multiple peripherals, it is necessary to derive many event objects of different lengths from QEvt. After obtaining the data, allocate memory through Q_NEW and send it to the receiving object via the post interface. Then, inside the receiving object, these data are copied again. This process will generate multiple buffering processes.
If dont use the Q_NEW API, I use global variables, Its simple but then I will be back to the situation where I need to manually protect resource. 
Could you please answer my question or provide a suitable example?

quantum-leaps, 2025-03-05:
Hi Andy,
Could you please repost your question to the [Free QP/QM Support Forum](https://sourceforge.net/p/qpc/discussion/668726/)? I want to answer your question, but such that this discussion will be available to the broader QP community. If I answer here, hardly anybody except you will see it. I hope this comment makes sense to you.
--MMS

andy27182, 2025-03-06:
sure, link: [Question about passing driver data with ...](https://sourceforge.net/p/qpc/discussion/668726/thread/7bbc437460/)